lowRISC/ibex-demo-system #16: Sysnthesis failure with Vivado 2022.2

jeras, 2022-12-29:
```
ERROR: [Synth 8-91] ambiguous clock in event control [/home/izi/VLSI/ibex-demo-system/build/lowrisc_ibex_demo_system_0/src/lowrisc_ibex_demo_system_core_0/rtl/system/gpio.sv:43]
```

The offending code is https://github.com/lowRISC/ibex-demo-system/blob/main/rtl/system/gpio.sv#L44
If the line was not supposed to have a reset, it should be placed into a separate always block without the reset edge in the sensitivity list. The current code seems to imply using reset as a clock.

GregAC, 2023-01-03:
Thanks for the report @jeras this will be addressed in https://github.com/lowRISC/ibex-demo-system/pull/13 which adds needed synchronization to the GPIO as well.

andreaskurth, 2023-01-05:
Closing as addressed in #13. If the issue persists, please reopen.

closed 2023-01-05 by andreaskurth