lowRISC/ibex-demo-system #40: Problem with loading hello world demo

AgamemnonasKyriazis, 2023-03-24:
I am executing the commands as said on the README and I'm getting the following error
after executing ./util/load_demo_system.sh run ./sw/build/demo/hello_world/demo
::
::

Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

closed 2023-04-04 by AgamemnonasKyriazis

wizkad, 2023-10-10:
I am getting a similar issue, what did you do to solve it?

AgamemnonasKyriazis, 2023-10-10:
I needed to change the configuration files that produced the bitstream. Didn't actually bother with it, just open the project using Vivado and change the board and reproduce the bitstream from there. Much easier and direct.

abhinav333, 2024-03-02:
Hello , I am facing a similar issue. Which configuration files are exactly required to be changed? 
My error is as follows:

Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Warn : JTAG tap: riscv.cpu       UNEXPECTED: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Error: JTAG tap: riscv.cpu  expected 1 of 1: 0x0362d093 (mfg: 0x049 (Xilinx), part: 0x362d, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: Debug Module did not become active. dmcontrol=0x0
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Thank you.

AgamemnonasKyriazis, 2024-03-02:
If my memory serves me right its the "ibex_demo_system.core" check whether the board used is the same as one you are trying to load the bitstream. There are many references on the board type, you may check every one of them.