lowRISC/ibex-demo-system #58: Unable to create a project file for synthesis.

abhinav333, 2023-07-21:
Hi, I am trying to replicate the ibex demo. Intend to create a Vivado project for FPGA synthesis for Arty board. Run fusesoc from Ibex core repository with the command:
`fusesoc --cores-root=. run --target=sim --tool=verilator --setup --build lowrisc:ibex:demo_system`
However get the following error log:
`WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_xnor2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_flop_en.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_and2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_usb_diff_rx.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_pad_attr.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_clock_inv.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_buf.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_flop.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_generic/prim_generic_xor2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_otp_pkg.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_usb_diff_rx.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_pad_wrapper.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_rom_pkg.core: 
Error validating data.filesets.files_rtl.depend must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lfsr.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_msb_extend.core: 
Error validating data.filesets.files_ascentlint_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_xnor2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_ram_1p_adv.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_mubi.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_pad_wrapper_pkg.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_ram_2p.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lc_sync.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_rom.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_buf.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_flop.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lc_and_hardened.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_xor2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_flop_2sync.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lc_or_hardened.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_rst_sync.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lc_dec.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_secded.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_flop_en.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_otp.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_clock_buf.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_and2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_pad_attr.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_clock_inv.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_flash.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_clock_gating.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_ram_1p_pkg.core: 
Error validating data.filesets.files_rtl.depend must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lc_combine.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_xoshiro256pp.core: 
Error validating data.filesets.files_ascentlint_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_ram_1p.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_blanker.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_clock_mux2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_ram_2p_pkg.core: 
Error validating data.filesets.files_rtl.depend must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim/prim_lc_sender.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_xilinx/prim_xilinx_and2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_xilinx/prim_xilinx_flop_en.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_xilinx/prim_xilinx_xor2.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_xilinx/prim_xilinx_flop.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
WARNING: Parse error. Ignoring file ./vendor/lowrisc_ip/ip/prim_xilinx/prim_xilinx_pad_attr.core: 
Error validating data.filesets.files_verilator_waiver.files must be array
ERROR: 'lowrisc:ibex:demo_system' or any of its dependencies requires 'demo_system', but this core was not found
`

rswarbrick, 2023-07-24:
My guess is that you're using a version of Fusesoc that doesn't understand some of our files. Can you make sure you've got the version listed in `python-requirements.txt`?

marnovandermaas, 2023-07-24:
Yes, we recommend following these instructions to setup your Python environment:
https://github.com/lowRISC/ibex-demo-system#native-python-environment

marnovandermaas, 2023-08-04:
I'm going to close this issue, but please reopen this issue or create a new issue if you have further questions.

closed 2023-08-04 by marnovandermaas

abhinav333, 2023-10-30:
Hi, I was able to create and synthesize Vivado project for Ibex demo system in Windows using the Anaconda environment with the suggestions you provided. However, now I am finding out how to program elf file on the running processor. However, we program the bit stream using a memory card and want to program the general vmem/elf alongside it.  No JTAG is available. Is there any method to do this? Thanks.