lowRISC/ibex-demo-system #59: zmmul extension issue

UnaiGalicia, 2023-07-24:
Hi, I am getting the following error when building the "demo" C program using lowrisc's toolchain (lowrisc-toolchain-rv32imcb-20230427-1 version):

``` /home/unai/…/lowrisc-toolchain-rv32imcb-20230427-1/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: -march=rv32i2p0_m2p0_c2p0_zmmul1p0: Invalid or unknown z ISA extension: 'zmmul' 
/home/unai/…/lowrisc-toolchain-rv32imcb-20230427-1/bin/../lib/gcc/riscv32-unknown-elf/10.2.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file ../../common/CMakeFiles/common.dir/demo_system.c.o  ```

However, this problem doesn't show up when building the same program with the official RISC-V GCC toolchain in my path. I assume that this is intended to work, as the program that I am trying to compile is the demo program (hello_world).

Thank you in advance!

nbdd0121, 2023-07-24:
I think our GCC version (released in 2020) is older than the zmmul extension. We might want to consider upgrading it? @luismarques

marnovandermaas, 2023-07-24:
Can you tell us what commands you are running when getting this error?
Are you following the instructions here:
https://github.com/lowRISC/ibex-demo-system#building-software

marnovandermaas, 2023-07-24:
I just tested those commands with `lowrisc-toolchain-rv32imcb-20230427-1` and everything is fine, so please do let us know how to replicate this error.

UnaiGalicia, 2023-07-24:
You are right, I reinstalled the toolchain and rerun the Python script for the dependencies and everything went fine too. I might have changed something that I wasn't supposed to. Sorry for the inconvenience...

closed 2023-07-24 by UnaiGalicia