lowRISC/ibex-demo-system #62: Aliasing problem of Python3 on Windows and unable to create new project file using Fusesoc on Windows

Shuvam-Saha, 2023-07-31:
Hello, I am attempting to clone the "ibex-demo-system" repository and set up a project on Xilinx Vivado for FPGA synthesis on the Arty Board. I am running "fusesoc" from the "ibex core" repository on the Anaconda command prompt. However, I encountered errors related to aliasing "python3" on Windows. Unlike other platforms, Windows saves executable files as "python.exe" by default instead of "python3". Could you kindly assist me in creating an alias for "python3" on Windows, so I can resolve these issues?

(ibex_env) D:\Users\DELL\Downloads\ibex_demo_system>fusesoc --cores-root=. run --target=synth --run lowrisc:ibex:demo_system
←[1;37mINFO: Preparing lowrisc:dv:crypto_prince_ref:0.1←[0m
←[1;37mINFO: Preparing lowrisc:dv:dv_fcov_macros:0←[0m
←[1;37mINFO: Preparing lowrisc:dv:dv_macros:0←[0m
←[1;37mINFO: Preparing lowrisc:dv:secded_enc:0←[0m
←[1;37mINFO: Preparing lowrisc:ibex:ibex_pkg:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:assert:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:cipher_pkg:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:primgen:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:ram_1p_pkg:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:ram_2p_pkg:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:secded:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:util_get_scramble_params:0←[0m
←[1;37mINFO: Preparing lowrisc:tool:check_tool_requirements:0.1←[0m
←[1;37mINFO: Preparing lowrisc:dv:scramble_model:0←[0m
←[1;37mINFO: Preparing lowrisc:dv_verilator:memutil_dpi:0←[0m
←[1;37mINFO: Preparing lowrisc:ibex:ibex_icache:0.1←[0m
←[1;37mINFO: Preparing lowrisc:ibex:rv_timer:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:cdc_rand_delay:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:cipher:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:lfsr:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:prim_pkg:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:util:0.1←[0m
←[1;37mINFO: Preparing lowrisc:dv_verilator:memutil_dpi_scrambled:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:buf:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:clock_gating:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:clock_inv:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:clock_mux2:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:flop:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:onehot_check:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:ram_1p:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:ram_2p:0←[0m
←[1;37mINFO: Preparing lowrisc:ibex:fpga_xilinx_shared:0←[0m
←[1;37mINFO: Preparing lowrisc:ibex:ibex_core:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:flop_2sync:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:ram_1p_adv:0.1←[0m
←[1;37mINFO: Preparing lowrisc:prim:fifo:0←[0m
←[1;37mINFO: Preparing lowrisc:prim:ram_1p_scr:0.1←[0m
←[1;37mINFO: Preparing lowrisc:ibex:ibex_top:0.1←[0m
←[1;37mINFO: Preparing pulp:riscv:debug_module:0←[0m
←[1;37mINFO: Preparing lowrisc:ibex:demo_system_core:0←[0m
←[1;37mINFO: Preparing lowrisc:ibex:demo_system:0←[0m
←[1;37mINFO: Generating lowrisc:prim:prim_pkg-impl:0.1←[0m
Traceback (most recent call last):
  File "D:\Users\DELL\Downloads\ibex_demo_system\vendor\lowrisc_ibex\vendor\lowrisc_ip\ip\prim\util/primgen.py", line 12, in <module>
    from mako.template import Template
ModuleNotFoundError: No module named 'mako'
←[1;31mERROR: Setup failed : "python3 D:\Users\DELL\Downloads\ibex_demo_system\vendor\lowrisc_ibex\vendor\lowrisc_ip\ip\prim\util/primgen.py D:\Users\DELL\Downloads\ibex_demo_system\build\lowrisc_ibex_demo_system_0\synth-vivado\generated\lowrisc_prim_prim_pkg-impl_0.1\impl_input.yml" exited with an error code. See stderr for details.←[0m

(ibex_env) D:\Users\DELL\Downloads\ibex_demo_system>
(ibex_env) D:\Users\DELL\Downloads\ibex_demo_system># If the above does not work, try executing the programming operation manually with..
'#' is not recognized as an internal or external command,
operable program or batch file.

(ibex_env) D:\Users\DELL\Downloads\ibex_demo_system>make -C ./build/lowrisc_ibex_demo_system_0/synth-vivado/ pgm
make: Entering directory 'D:/Users/DELL/Downloads/ibex_demo_system/build/lowrisc_ibex_demo_system_0/synth-vivado'
make: *** No rule to make target 'pgm'.  Stop.
make: Leaving directory 'D:/Users/DELL/Downloads/ibex_demo_system/build/lowrisc_ibex_demo_system_0/synth-vivado'

marnovandermaas, 2023-08-01:
We don't officially support Windows or Anaconda, but we have seen people have some luck running the demo system under Windows subsystem for Linux. You can also run a docker image, you may find some hints on how to run things under Windows here:
https://github.com/lowRISC/ibex-demo-system-labs/blob/main/lab1.md
Most of our development happens on Linux, so if everything fails you can try running Linux in a virtual machine.

marnovandermaas, 2023-08-04:
I'm closing this issue, but please reopen it or open another issue if you have more questions.

closed 2023-08-04 by marnovandermaas