lowRISC/ibex-demo-system #77: Building Simulation

utku1998, 2023-12-18:
When I try to build the Simulator for "Ibex Demo System", using the command:
"**fusesoc --cores-root=. run --target=sim --tool=verilator --setup --build lowrisc:ibex:demo_system**".

I got the following error mesage:
"""
INFO: Running pre_build script check_tool_requirements
INFO: Building simulation model
INFO: verilator -f lowrisc_ibex_demo_system_0.vc -DDISABLE_PRIM_CDC_RAND_DELAY --trace --trace-fst --trace-structs --trace-params --trace-max-array 1024 -CFLAGS "-std=c++11 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=ibex_demo_system" -LDFLAGS "-pthread -lutil -lelf" -Wall -Wwarn-IMPERFECTSCH --unroll-count 72

ERROR: %Warning-GENUNNAMED: ../src/lowrisc_ibex_demo_system_core_0/rtl/system/ibex_demo_system.sv:475:12: Unnamed generate block 'genblk2' (IEEE 1800-2017 27.6)
                                                                                                 : ... Suggest assign a label with 'begin : gen_<label_name>'
  475 |   end else begin
      |            ^~~~~
                     ... For warning description see https://verilator.org/warn/GENUNNAMED?v=5.018
                     ... Use "/* verilator lint_off GENUNNAMED */" and lint_on around source to disable this message.
%Warning-GENUNNAMED: ../src/lowrisc_ibex_demo_system_core_0/rtl/system/gpio.sv:39:3: Unnamed generate block 'genblk1' (IEEE 1800-2017 27.6)
                                                                                   : ... Suggest assign a label with 'begin : gen_<label_name>'
   39 |   for (genvar i = 0; i < GpiWidth; i++) begin
      |   ^~~
%Warning-GENUNNAMED: ../src/lowrisc_ibex_demo_system_core_0/rtl/system/spi_host.sv:118:15: Unnamed generate block 'genblk1' (IEEE 1800-2017 27.6)
                                                                                         : ... Suggest assign a label with 'begin : gen_<label_name>'
  118 |     if (CPHA) begin
      |               ^~~~~
%Warning-GENUNNAMED: ../src/lowrisc_ibex_demo_system_core_0/rtl/system/spi_host.sv:139:14: Unnamed generate block 'genblk1' (IEEE 1800-2017 27.6)
                                                                                         : ... Suggest assign a label with 'begin : gen_<label_name>'
  139 |     end else begin
      |              ^~~~~
%Error: Exiting due to 4 warning(s)
make: *** [Makefile:16: Vibex_demo_system.mk] Error 1

ERROR: Failed to build lowrisc:ibex:demo_system:0 : '['make', '-j', '4']' exited with an error: 2

"""

Does anyone know the reason and have a solution? I would really appreciate it, thanks :)!

marnovandermaas, 2023-12-18:
Hmmm, not sure if I've seen that issue before. Can you tell me what version of Verilator you're running?

This is the one I currently use:
```
$ verilator --version
Verilator 4.210 2021-07-07 rev v4.210
```

utku1998, 2023-12-18:
Thanks for your fast response!

I am using the following version:

```
$ verilator --version
Verilator 5.018 2023-10-30 rev v5.018-38-g344f87abe
```

utku1998, 2023-12-18:
@marnovandermaas do you have any suggestion other than changing the verilator version (I assume it works on yours!)?

Because with this version "Ibex Simple System" example works just fine. Therefore, I would like to stay with this version, if possible.

utku1998, 2023-12-18:
@marnovandermaas, to solve this issue, i have disabled the warnings using "**/* verilator lint_off GENUNNAMED */**" before the blocks and "**/* verilator lint_on GENUNNAMED */**" afterwards. 

This has solved the problem and I could build the simulation:
[image]

But I would appreciate to learn if there is a better solution.

marnovandermaas, 2023-12-18:
We can add the labels as it suggests. I'll create a PR for this.

utku1998, 2023-12-18:
Thanks, will be waiting for it!

marnovandermaas, 2023-12-18:
@utku1998 can you check whether the code here works for you? https://github.com/lowRISC/ibex-demo-system/pull/79

utku1998, 2023-12-19:
@marnovandermaas Yes, now it works. Thanks!

closed 2023-12-21 by marnovandermaas