lowRISC/ibex-demo-system #78: Building Software

utku1998, 2023-12-18:
Hello, I have executed the command ```cmake ..``` successfully.

However, I once execute the make command, the process will be executed till [100%], but then I got the following error:

"""
[ 93%] Linking C executable basic-passwdcheck
[ 93%] Built target basic-passwdcheck
Scanning dependencies of target blank
[ 96%] Building ASM object blank/CMakeFiles/blank.dir/blank.S.obj
[100%] Linking ASM executable blank
/usr/bin/objcopy: Unable to recognise the format of the input file `path_to_repo/ibex-demo-system/sw/c/build/blank/blank'
make[2]: *** [blank/CMakeFiles/blank.dir/build.make:97: blank/blank] Error 1
make[2]: *** Deleting file 'blank/blank'
make[1]: *** [CMakeFiles/Makefile2:421: blank/CMakeFiles/blank.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
"""

I can see the binaries of all demos (basic-passwdcheck, hello_world. lcd_st7735, simpleserial-aes) in their respective folders. But what is the meaning of the last error and is it important?

I would appreciate, if someone knowing the reason of this issue, can explain.

Thanks in advance!

marnovandermaas, 2023-12-18:
I've never seen this error before. Can you tell me whether it generated the binary: `./sw/c/build/blank/blank`? Did you install your RISC-V toolchain from here: https://github.com/lowRISC/lowrisc-toolchains/releases?

utku1998, 2023-12-18:
Thanks for your fast response @marnovandermaas. No, there is no "blank" in the "/sw/c/build/blank/blank":
[image]
What is the purpose of it?

Although I got previosly mentioned error, I could still run the "hello_world" software on the simulated system:
[image]

That's why, I have asked what is the purpose of this "blank" and what problem may this error bring?

Aaronyap2002, 2023-12-20:
hi! I am having trouble building the software as well and posted an issuw #82. Would you be kind to let me know how did u get `cmake` to run?

virajwadhwa, 2024-04-20:
Having the same issue!!!

marnovandermaas, 2024-04-22:
The purpose of the `blank` binary is for programming the FPGA. It is essentially a software image with all zeros. It looks like it is using the wrong `objcopy` on your system.  Can you try the following patch:
```
diff --git a/sw/c/gcc_toolchain.cmake b/sw/c/gcc_toolchain.cmake
index d84d1af..00d9e24 100644
--- a/sw/c/gcc_toolchain.cmake
+++ b/sw/c/gcc_toolchain.cmake
@@ -1,6 +1,7 @@
 set(LINKER_SCRIPT "${CMAKE_CURRENT_LIST_DIR}/../common/link.ld")
 set(CMAKE_SYSTEM_NAME Generic)
 set(CMAKE_C_COMPILER riscv32-unknown-elf-gcc)
+set(CMAKE_OBJCOPY riscv32-unknown-elf-objcopy)
 set(CMAKE_C_FLAGS_INIT
     "-march=rv32imc -mabi=ilp32 -mcmodel=medany -Wall -fvisibility=hidden -
ffreestanding")
 set(CMAKE_ASM_FLAGS_INIT "-march=rv32imc")
```

marnovandermaas, 2024-04-22:
Please feel free to reopen the issue if the fix does not work.

closed 2024-04-22 by marnovandermaas