lowRISC/ibex-demo-system #80: A basic peripheral to write ASCII output to a file and halt simulation from software

utku1998, 2023-12-19:
In the documentation it is written that there is a basic peripheral to write ASCII output to a file and halt simulation from software. I assume that this is only used in case of a Verilator right? In case of a real target, FPGA, the ASCII output will be sent to the UART and not written to the ".log" file.

[image]

Functions like puts(), puthex(), putchar() work perfectly fine for the "Ibex Simple System (https://github.com/lowRISC/ibex/blob/master/examples/simple_system/README.md)" when simulating it on the Verilator. The output is written to file "ibex_simple_system.log":

[image]

However, the same functions do not write to the file "ibex_demo_system.log", which is still empty after the simulation. Am I doing or understanding wrong or is there a issue there?

[image]

Thanks in advance!

marnovandermaas, 2023-12-19:
This is not something I have tested before. There is a `SIM_CTRL_OUTPUT` flag for `cmake`. Can you try the following command and report back?
`$ cmake -DSIM_CTRL_OUTPUT:BOOL=ON ..`

utku1998, 2023-12-19:
Yes, it worked perfectly. Thanks!