lowRISC/ibex-demo-system #81: Missing System Memory Map for Ibex Demo System

utku1998, 2023-12-19:
In the case of "Ibex Simple System (https://github.com/lowRISC/ibex/tree/master/examples/simple_system)", the memory map of the system is shown in the figure below: 

[image]

However, I could not find a similar memory map, which shows where are the memory regions of the "Ibex Demo System" peripherals, such as GPIO, SPI, UART and also RAM etc. In the file "demo_system_regs.h", I have found the following values:

[image]

But is there a memory map (documentation), which shows the exact memory regions (start address, size, etc.) of the peripherals and RAM?

marnovandermaas, 2023-12-19:
Unfortunately, we don't have documentation for this. But we do have the software abstractions to deal with this. For example, if you want to interact with SPI you can include https://github.com/lowRISC/ibex-demo-system/blob/main/sw/c/common/spi.h
This also has the register offsets in it. However, you should not need these as you can use the functions to interact with the block:
```c
void spi_init(spi_t *spi, spi_reg_t reg, uint32_t speed);
void spi_send_byte_blocking(spi_t *spi, char c);
spi_status_t spi_get_status(spi_t *spi);
```

Additionally, you can find the offsets and size of regions in the RTL: https://github.com/lowRISC/ibex-demo-system/blob/main/rtl/system/ibex_demo_system.sv#L32-L63

marnovandermaas, 2025-04-11:
Closing this due to inactivity, feel free to re-open if you have further enquiries.