lowRISC/ibex-demo-system #82: Can't build software, cmake file not found

Aaronyap2002, 2023-12-20:
Hi, i was setting the system up, succesfully got docker to setup...
```aron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system$ sudo docker run -it --rm \
>   -p 6080:6080 \
>   -p 3333:3333 \
>   -v $(pwd):/home/dev/demo:Z \
>   ibex
[sudo] password for aaron: 
Please be aware that you are exposing your VNC server to all users on the
local machine. These users can access your server without authentication!
/usr/bin/xauth:  file /home/dev/.Xauthority does not exist

New '33f8941bd738:1 (dev)' desktop at :1 on machine 33f8941bd738

Starting applications specified in startxfce4
Log file is /home/dev/.vnc/33f8941bd738:1.log

Use xtigervncviewer -SecurityTypes None :1 to connect to the VNC server.

WebSocket server settings:
  - Listen on :6080
  - Web server. Web root: /usr/share/novnc
  - SSL/TLS support
  - proxying from :6080 to localhost:5901
172.17.0.1 - - [20/Dec/2023 07:34:50] 172.17.0.1: Plain non-SSL (ws://) WebSocket connection
172.17.0.1 - - [20/Dec/2023 07:34:50] 172.17.0.1: Path: '/websockify'
172.17.0.1 - - [20/Dec/2023 07:34:50] connecting to: localhost:5901
```
My issue is that, i could not run this set of commands.
```cd /home/dev/demo
mkdir -p sw/build
pushd sw/build
cmake ..
make
popd
```
It produced this set of errors, 
```aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system$ cd /home/dev/demo

cmake ..
make
popdbash: cd: /home/dev/demo: No such file or directory
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system$ mkdir -p sw/build
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system$ pushd sw/build
~/Downloads/ibex-demo-system/sw/build ~/Downloads/ibex-demo-system ~/Downloads/ibex-demo-system/sw/build ~/Downloads/ibex-demo-system ~ ~
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system/sw/build$ cmake ..
CMake Error: The source directory "/home/aaron/Downloads/ibex-demo-system/sw" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system/sw/build$ make
make: *** No targets specified and no makefile found.  Stop.
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system/sw/build$ popd
~/Downloads/ibex-demo-system ~/Downloads/ibex-demo-system/sw/build ~/Downloads/ibex-demo-system ~ ~
```
I not sure what is `dev` in the `cd /home/dev/demo` command suppose to be.  There is no cmake file in `sw/build` 
```
aaron@aaron-TravelMate-P238-M:~/Downloads/ibex-demo-system/sw$ ls
build  c  common  rust
```

marnovandermaas, 2023-12-20:
Those instructions are out of date. You need to make this directory: `sw/c/build` with the `c`. You should be able to find the instructions here:
https://github.com/lowRISC/ibex-demo-system#c-stack

Where did you find these old build instructions?

Aaronyap2002, 2023-12-20:
i found it here. I assume this whole repository could not be use again? [https://github.com/lowRISC/ibex-demo-system-labs/blob/main/lab1.md](url)

marnovandermaas, 2024-04-22:
Yes, you can skip those instructions if you don't have an FPGA. Those labs were written with an old version of the demo system: https://github.com/lowRISC/ibex-demo-system/releases/tag/v0.0.3

closed 2024-04-22 by marnovandermaas