lowRISC/ibex-demo-system #89: Can't build docker Container

tju-sun-lab, 2024-01-20:
=> [internal] load build definition from Dockerfile                                                                                                  0.0s
 => => transferring dockerfile: 3.88kB                                                                                                                0.0s
 => [internal] load .dockerignore                                                                                                                     0.0s
 => => transferring context: 2B                                                                                                                       0.0s
 => [internal] load metadata for docker.io/library/ubuntu:20.04                                                                                       2.0s
 => [ 1/17] FROM docker.io/library/ubuntu:20.04@sha256:f2034e7195f61334e6caff6ecf2e965f92d11e888309065da85ff50c617732b8                               0.0s
 => [internal] load build context                                                                                                                     0.0s
 => => transferring context: 543B                                                                                                                     0.0s
 => CACHED [ 2/17] RUN ln -sf /bin/bash /bin/sh                                                                                                       0.0s
 => CACHED [ 3/17] RUN OBS_URL="https://download.opensuse.org/repositories";     OBS_PATH="/home:/phiwag:/edatools/xUbuntu_20.04";     REPO_URL="${O  0.0s
 => CACHED [ 4/17] COPY container/apt-requirements.txt /tmp/apt-requirements.txt                                                                      0.0s
 => CACHED [ 5/17] RUN echo "verilator-4.210" >> /tmp/apt-requirements.txt     && sed -i -e '/^$/d' -e '/^#/D' -e 's/#.*//' /tmp/apt-requirements.tx  0.0s
 => CACHED [ 6/17] COPY container/get-toolchain.py /tmp/get-toolchain.py                                                                              0.0s
 => ERROR [ 7/17] RUN /tmp/get-toolchain.py -r 20220210-1     && rm -f /tmp/get-toolchain.py                                                          0.2s
------                                                                                                                                                     
 > [ 7/17] RUN /tmp/get-toolchain.py -r 20220210-1     && rm -f /tmp/get-toolchain.py:
0.180 /usr/bin/env: 'python3': No such file or directory
------
Dockerfile:57
--------------------
  56 |     COPY container/get-toolchain.py /tmp/get-toolchain.py
  57 | >>> RUN /tmp/get-toolchain.py -r ${RISCV_TOOLCHAIN_RELEASE_TAG} \
  58 | >>>     && rm -f /tmp/get-toolchain.py
  59 |     ENV PATH "/tools/riscv/bin:${PATH}"
--------------------
ERROR: failed to solve: process "/bin/sh -c /tmp/get-toolchain.py -r ${RISCV_TOOLCHAIN_RELEASE_TAG}     && rm -f /tmp/get-toolchain.py" did not complete successfully: exit code: 127

marnovandermaas, 2024-01-22:
I just tested this command on my Ubuntu 20 machine from the root of the repo:
`sudo docker build . -t ibex -f container/Dockerfile`

This successfully build a Docker image for me. Can you confirm that you ran the same command? Unfortunately I don't have a Mac machine available for me to test this on.

I also see that it is pulling a cached version for step 5/17 in your build log, which is where python3 gets installed. Do you have an earlier version of the log to check whether python3 is successfully installed by apt?

Also can you tell me the version of Docker that you are using? This is mine:
```
$ docker --version
Docker version 24.0.5, build 24.0.5-0ubuntu1~20.04.1
```

tju-sun-lab, 2024-01-27:
> I've also made a release that includes a docker image in case you want to skip the building phase: https://github.com/lowRISC/ibex-demo-system/releases/tag/v0.0.3

Thank you very much! That works for me :D

closed 2024-04-22 by marnovandermaas