lowRISC/ibex-demo-system #90: Unable to run simulator for Hello World demo due to missing files

Aaronyap2002, 2024-01-30:
hi everyone. As I was following the instructions to run the simulator, when I tried to run the command `./build/lowrisc_ibex_demo_system_0/sim-verilator/Vibex_demo_system \
  --meminit=ram,./sw/c/build/demo/hello_world/demo`, it produced the following errors, and I don't know how to fix the issues. My verilator version is 4.210. 

```ERROR: In file included from ../src/lowrisc_ibex_demo_system_0/dv/verilator/ibex_demo_system.cc:9:
./Vibex_demo_system__Syms.h:10:10: fatal error: verilated_heavy.h: No such file or directory
   10 | #include "verilated_heavy.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Vibex_demo_system.mk:120: ibex_demo_system.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from ../src/lowrisc_dv_verilator_simutil_verilator_0/cpp/verilated_toplevel.h:21,
                 from ../src/lowrisc_ibex_demo_system_0/dv/verilator/ibex_demo_system.h:5,
                 from ../src/lowrisc_ibex_demo_system_0/dv/verilator/ibex_demo_system_main.cc:5:
./Vibex_demo_system.h:11:10: fatal error: verilated_heavy.h: No such file or directory
   11 | #include "verilated_heavy.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Vibex_demo_system.mk:122: ibex_demo_system_main.o] Error 1
In file included from ../src/lowrisc_dv_verilator_simutil_verilator_0/cpp/verilated_toplevel.h:21,
                 from ../src/lowrisc_dv_verilator_simutil_verilator_0/cpp/verilated_toplevel.cc:5:
./Vibex_demo_system.h:11:10: fatal error: verilated_heavy.h: No such file or directory
   11 | #include "verilated_heavy.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Vibex_demo_system.mk:116: verilated_toplevel.o] Error 1
In file included from ../src/lowrisc_dv_verilator_simutil_verilator_0/cpp/verilated_toplevel.h:21,
                 from ../src/lowrisc_dv_verilator_simutil_verilator_0/cpp/verilator_sim_ctrl.h:13,
                 from ../src/lowrisc_dv_verilator_simutil_verilator_0/cpp/verilator_sim_ctrl.cc:5:
./Vibex_demo_system.h:11:10: fatal error: verilated_heavy.h: No such file or directory
   11 | #include "verilated_heavy.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Vibex_demo_system.mk:118: verilator_sim_ctrl.o] Error 1
make: *** [Makefile:13: Vibex_demo_system] Error 2

ERROR: Failed to build lowrisc:ibex:demo_system:0 : '['make', '-j', '4']' exited with an error: 2
```

marnovandermaas, 2024-02-01:
`verilated_heavy.h` should be located in the standard Verilator include directory. For me this is located close to my verilator binary.
You can find where you Verilator binary is by executing: `which verilator`
From this path I can remove `bin/verilator` and add `share/verilator/include`. There resulting directory contains `veraltored_heavy.h`.

I'm not exactly sure how you installed Verilator, but I assume something went wrong in the process. I recommend that you follow the "Git quick install" instructions here: https://verilator.org/guide/latest/install.html#git-quick-install

Aaronyap2002, 2024-02-01:
```
diff --git a/sw/c/gcc_toolchain.cmake b/sw/c/gcc_toolchain.cmake index d84d1af..00d9e24 100644 --- a/sw/c/gcc_toolchain.cmake +++ b/sw/c/gcc_toolchain.cmake @@ -1,6 +1,7 @@ set(LINKER_SCRIPT "${CMAKE_CURRENT_LIST_DIR}/../common/link.ld") set(CMAKE_SYSTEM_NAME Generic) set(CMAKE_C_COMPILER riscv32-unknown-elf-gcc) +set(CMAKE_OBJCOPY riscv32-unknown-elf-objcopy) set(CMAKE_C_FLAGS_INIT "-march=rv32imc -mabi=ilp32 -mcmodel=medany -Wall -fvisibility=hidden -ffreestanding") set(CMAKE_ASM_FLAGS_INIT "-march=rv32imc")
```
How should I run this patch? Just run it in the terminal right?

rswarbrick, 2024-02-02:
Marno is suggesting you apply the patch to `gcc_toolchain.cmake`. There is probably a way to do it with the `patch` command, but the easiest way would be to manually edit that file to add the line in question.

marnovandermaas, 2024-04-22:
I've opened up a PR for this: https://github.com/lowRISC/ibex-demo-system/pull/107

marnovandermaas, 2024-04-22:
Closing this as it is a duplicate of: https://github.com/lowRISC/ibex-demo-system/issues/78

closed 2024-04-22 by marnovandermaas