lowRISC/ibex-demo-system #95: Error while building simulation

NagaHimanshu, 2024-02-16:
While building simulation from Ibex Repo folder, I got this error.

_sh-5.0$ fusesoc --cores-root=. run --target=sim --tool=verilator --setup --build lowrisc:ibex:demo_system
ERROR: 'lowrisc:ibex:demo_system' or any of its dependencies requires 'demo_system', but this core was not found_

How to fix this issue?

NagaHimanshu, 2024-02-17:
Thanks for the comment. Since in the instructions, was mentioned to run "from Ibex repository root", I thought of running from this root - https://github.com/lowRISC/ibex.  I guess that was the fault. 

In Linux environment, I built the Software and used the docker - https://github.com/lowRISC/ibex-demo-system/releases/download/v0.0.3/ibex-demo-system-docker.tar.gz. After connecting to server and if I execute that command, I'm getting this error:

_INFO: Wrote dependency graph to /home/dev/demo/build/lowrisc_ibex_demo_system_0/sim-verilator/lowrisc_ibex_demo_system_0.deps-after-generators.dot
ERROR: Setup failed : Cannot find lint/verilator_waiver.vlt in :
	.
	._

 Can you help me with this?

marnovandermaas, 2025-04-11:
Yup, I don't get that error when I build the simulator. Can you try reproducing this on the latest main?

Other things you can try is checking your Verilator version:
```sh
❯ verilator --version
Verilator 5.024 2024-04-05 rev v5.024
```

And looking for the vlt file:
```sh
❯ find -name "verilator_waiver.vlt"
./build/lowrisc_ibex_demo_system_0/src/lowrisc_ibex_ibex_core_0.1/lint/ve
rilator_waiver.vlt
./build/lowrisc_ibex_demo_system_0/src/lowrisc_ibex_ibex_top_0.1/lint/ver
ilator_waiver.vlt
./vendor/lowrisc_ibex/examples/simple_system/lint/verilator_waiver.vlt
./vendor/lowrisc_ibex/lint/verilator_waiver.vlt
./vendor/lowrisc_ibex/dv/cs_registers/lint/verilator_waiver.vlt
./vendor/lowrisc_ibex/dv/riscv_compliance/lint/verilator_waiver.vlt
```