lowRISC/ibex-demo-system #98: Corrupted lowrisc_prim_abstract_xx RTL files

NagaHimanshu, 2024-02-22:
Hi, I am building the Ibex demo system in windows and successfully built the software. While running the command for synthesis

`fusesoc --cores-root=. run --target=synth --setup --build lowrisc:ibex:demo_system`

The output from command-line is as in the attached file - [cmd_output.txt](https://github.com/lowRISC/ibex-demo-system/files/14369241/cmd_output.txt)

This generated the prim_xx.sv files that are corrupted and providing syntax errors. I am attaching two of such generated files (uploading copied txt files since sv files cannot be uploaded) - 
[prim_and2.txt](https://github.com/lowRISC/ibex-demo-system/files/14369249/prim_and2.txt)
[prim_buf.txt](https://github.com/lowRISC/ibex-demo-system/files/14369912/prim_buf.txt)

Could you please assist me with this issue?

NagaHimanshu, 2024-02-22:
I found the issue, that, the files generated using verible parser created problem whereas, that of regex parser are proper. Is there any way, that by default to bypass the verible parser without much modification to the code?

marnovandermaas, 2024-02-22:
Yes, those files don't look good to me.

Here's an example of my generated `prim_and2.sv`:
```SystemVerilog
// Copyright lowRISC contributors.
// Licensed under the Apache License, Version 2.0, see LICENSE for details.
// SPDX-License-Identifier: Apache-2.0

// This file is auto-generated.
// Used parser: Verible

`ifndef PRIM_DEFAULT_IMPL
  `define PRIM_DEFAULT_IMPL prim_pkg::ImplGeneric
`endif

// This is to prevent AscentLint warnings in the generated
// abstract prim wrapper. These warnings occur due to the .*
// use. TODO: we may want to move these inline waivers
// into a separate, generated waiver file for consistency.
//ri lint_check_off OUTPUT_NOT_DRIVEN INPUT_NOT_READ HIER_BRANCH_NOT_READ
module prim_and2

#(
parameter int Width = 1
) (
  input        [Width-1:0] in0_i,
  input        [Width-1:0] in1_i,
  output logic [Width-1:0] out_o
);
  parameter prim_pkg::impl_e Impl = `PRIM_DEFAULT_IMPL;

if (Impl == prim_pkg::ImplXilinx) begin : gen_xilinx
    prim_xilinx_and2 #(
      .Width(Width)
    ) u_impl_xilinx (
      .*
    );
end else begin : gen_generic
    prim_generic_and2 #(
      .Width(Width)
    ) u_impl_generic (
      .*
    );
end

endmodule
//ri lint_check_on OUTPUT_NOT_DRIVEN INPUT_NOT_READ HIER_BRANCH_NOT_READ
```

marnovandermaas, 2024-02-22:
Are you running on Windows Subsystem for Linux?

marnovandermaas, 2024-02-22:
There may be some hints on running the demo system on Windows in the labs:
https://github.com/lowRISC/ibex-demo-system-labs/blob/main/lab1.md#windows
https://github.com/lowRISC/ibex-demo-system-labs/blob/main/lab1.md#set-up-container-windows

NagaHimanshu, 2024-02-22:
I am building it for Windows using command-line, not in WSL.

marnovandermaas, 2025-04-11:
Windows command line is unlikely to work. I recommend switching to WSL.