lowRISC/ibex-demo-system #99: Cannot run the Docker container on Windows following the instructions in the main readme file

Sam-Vervaeck, 2024-02-25:
First of all, I had to clear a smaller issue to find the container in the first place. This instruction in the readme says there is a prebuilt container ready, but then it does not mention where it is located:

> There is a prebuilt container of tools available you may want to use to get started quickly. There are instructions for building the container for either Docker/Podman located in ./container/README.md.

By looking through the issues and going to the releases I managed to find a .tar.gz container in the latest release. I downloaded it and extracted the .tar file into my ibex-demo-system local cloned repository.

Next, I ran these commands from the Windows section in an administrator cmd:

`cd "C:\Program Files\Docker\Docker"
.\DockerCli.exe -SwitchLinuxEngine`

Then, this instruction from the readme is not very clear on where to run the next commands, so i changed directories back to the ibex-demo-system directory:

> In case you have a tarball of the docker image, run:

Finally, I ran this code with the name of the .tar file changed to the one from the release:

`docker load -i ibex-demo-system-docker.tar`

And that gave me the following error:

`error during connect: in the default daemon configuration on Windows, the docker client must be run with elevated privileges to connect: Post "http://%2F%2F.%2Fpipe%2Fdocker_engine/v1.24/images/load?quiet=0": open //./pipe/docker_engine: The system cannot find the file specified.`

I find it weird that it asks about elevated privileges as i am in an administrator cmd window.

marnovandermaas, 2024-02-26:
I don't have a Windows machine to test this on. Have you tried using the Docker Desktop GUI instead?

Sam-Vervaeck, 2024-02-26:
> I don't have a Windows machine to test this on. Have you tried using the Docker Desktop GUI instead?

I have followed the docker desktop gui setup and arrived at this when running the container release:
[image]

Navigating to http://localhost:6080/vnc.html in Chrome browser results in a "connected (unencrypted)..." message but then the vnc windows stays black...