lowRISC/ibex-demo-system #102: The pip install setup for the container seems incomplete as fusesoc throws an error on a clean run.

Sam-Vervaeck, 2024-03-05:
When creating the container fresh after cloning this repository and when into the VNC viewer, running the fusesoc command:
`fusesoc --cores-root=. run --target=sim --tool=verilator --setup --build lowrisc:ibex:demo_system`
results in an error:
`ERROR: Traceback (most recent call last):
  File "util/check_tool_requirements.py", line 10, in <module>
    from packaging.version import Version
ModuleNotFoundError: No module named 'packaging'

ERROR: Failed to build lowrisc:ibex:demo_system:0 : pre_build script 'check_tool_requirements': ['python3', 'util/check_tool_requirements.py'] exited with error code 1`

After running pip install packaging the fusesoc command runs without errors

marnovandermaas, 2024-03-05:
I think you're right. We need to add `packaging` to https://github.com/lowRISC/ibex-demo-system/blob/main/python-requirements.txt
Would you like to contribute this with a pull request? I'm happy to do it otherwise.

Sam-Vervaeck, 2024-03-07:
I'm not very familiar with the python requirements flow, would have to figure it out first.

Sam-Vervaeck, 2024-08-09:
I see that it has been resolved in a commit.

closed 2024-08-09 by Sam-Vervaeck

marnovandermaas, 2024-08-12:
Thanks for following up on this issue Sam!