lowRISC/ibex-demo-system #105: Issue with Vtop_verilator when running simulator. ; Replaced with Vibex_demo_system

Aaronyap2002, 2024-04-07:
Hello, so as i was going through the procedures, i was trying to run the simulator by running `./build/lowrisc_ibex_demo_system_0/sim-verilator/Vtop_verilator \ --meminit=ram,./sw/c/build/demo/hello_world/demo `, as stated. However, for some reason it produced this error:
`./build/lowrisc_ibex_demo_system_0/sim-verilator/Vtop_verilator: No such file or directory
`
So what i did was i went to the `sim_verilator` directory to find `Vtop_verilator`. Then i found that there is an executable file named `Vibex_demo_system`. I replaced `Vtop_verilator` with `Vibex_demo_system`, and re-run the command. 
Then it worked as intended. I would like to know if this is normal? Or there is a mix-up somewhere?

marnovandermaas, 2024-04-08:
This changed about 2 months ago, can you check that you have checked out the latest commit. I tested this on commit 79b4f8176198f3492e64d595fb8c08f2aeb28deb

marnovandermaas, 2025-04-11:
Closing this issue because of inactivity, please feel free to re-open if you have further enquiries.

closed 2025-04-11 by marnovandermaas