lowRISC/ibex-demo-system #111: Issues with using timer

newfrogg, 2024-05-13:
Hi all, I'm using the hello_world demo. I'm want to print a pattern 1 times inside the main while loop, but the pattern is printed multiple times inside the loop. Below is my code. Any help will be appreciated. Thanks. 
P/S: issue is about printing pattern drived by the value of flag: 

```
      if (flag == 0) {
        // write_r_valid(1);
        // puts("\n");
        flag = 1;
        puts("/************/");
        // puthex(flag);
      }
      else {
        puts("/**_____**/");
      }
```

```
int8_t flag = 0;

int main(void) {
  install_exception_handler(UART_IRQ_NUM, &test_uart_irq_handler);

  uart_enable_rx_int();
  // This indicates how often the timer gets updated.
  timer_init();
  // timer_enable(5000000);
  // timer_enable(4000000);
  timer_enable(10000000);

  uint64_t last_elapsed_time = get_elapsed_time();

  // Reset green LEDs to having just one on
  set_outputs(GPIO_OUT, 0x10);  // Bottom 4 bits are LCD control as you can see in top_artya7.sv
  while (1) {
    uint64_t cur_time = get_elapsed_time();

    if (cur_time != last_elapsed_time) {
      last_elapsed_time = cur_time;

      // Disable interrupts whilst outputting to prevent output for RX IRQ
      // happening in the middle

      set_global_interrupt_enable(0);

      uint32_t in_val = read_gpio(GPIO_IN_DBNC);

      puts(" [OUTPUT] r_data: ");
      r_data = read_r_data();
      puthex(r_data);

      puts(" [OUTPUT] w_valid: ");
      w_valid = read_w_valid();
      puthex(w_valid);

      puts(" [OUTPUT] t_valid: ");
      t_valid = read_t_valid();
      puthex(t_valid);

      puts(" [OUTPUT] out_data: ");
      out_data = read_r_data(H_t_OUT_X4_1);
      puthex(out_data);
      puts("\n");

      if (flag == 0) {
        // write_r_valid(1);
        // puts("\n");
        flag = 1;
        puts("/************/");
        // puthex(flag);
      }
      else {
        puts("/**_____**/");
      }

      // Re-enable interrupts with output complete
      set_global_interrupt_enable(1);
      /*
        Display PWM - for testing the timer frequency
      */
      {
        // Cycling through green LEDs
        if (USE_GPIO_SHIFT_REG) {
          // Feed value of BTN0 into the shift register
          set_outputs(GPIO_OUT_SHIFT, in_val);
        } else {
          // Cycle through LEDs unless BTN0 is pressed
          uint32_t out_val = read_gpio(GPIO_OUT);
          out_val          = (out_val << 1) & GPIO_LED_MASK;
          if ((in_val & 0x1) || (out_val == 0)) {
            out_val = 0x10;
          }
          set_outputs(GPIO_OUT, out_val);
        }
      }
    }

  }
}
```

Here the output (printed UART)
[image]

marnovandermaas, 2024-05-15:
I'm not sure how well globals work in the current setup. Can you try to put the `flag` variable on the stack:
```c
int main(void) {
  int flag = 0;
  ...
  while (1) {
    ...
  }
}

marnovandermaas, 2025-04-11:
Closing this issue because of inactivity, please feel free to re-open if you have further enquiries.

closed 2025-04-11 by marnovandermaas