lowRISC/ibex-demo-system #121: "prim_ram_2p.sv" not found

rezot98, 2024-09-09:
Hello, I am trying to build the project targeter on artya7 board, manualy (Without using FuseSoC) for better understanding. I found out that the file "ram_2p.sv" instantiates a module "prim_ram_2p" that can not be found anywhere. In addition, there were some other files that were instantiated but not found anywhere. Regardless of those issues the verilator simulation was run without any errors. I would like to know why there are no "prim_ram_2p.sv" and how did the verilator simulation still run? Thanks

marnovandermaas, 2024-09-16:
Fusesoc is currently being used to generate these primitives. I recommend running Fusesoc to generate those files and then run your alternative synthesis flow once those files have been created. We don't currently support running any other flow besides FuseSoC in this repository.

marnovandermaas, 2025-04-11:
Closing this issue because of inactivity, please feel free to re-open if you have further enquiries.

closed 2025-04-11 by marnovandermaas