lowRISC/ibex-demo-system #122: Unable to load a firmware if the size is bigger than a certain value

Clay-Cui, 2024-09-12:
So I'm stuck when trying to load big(relatively) firmware/program to the ibex-demo-system. After several trial, I find that if the program file is smaller than 9408 bytes as shown below (9404B), it will work. It can print out through uart and everything work. 
````
./util/load_demo_system.sh cw305 run ./sw/c/build/demo/aes/aes 
Open On-Chip Debugger 0.12.0-rc1
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Warn : 'riscv set_prefer_sba' is deprecated. Please use 'riscv set_mem_access' instead.
force hard breakpoints
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
9404 bytes written at address 0x00100000
downloaded 9404 bytes in 0.226321s (40.578 KiB/s)

Error: No working memory available. Specify -work-area-phys to target.
Warn : not enough working area available(requested 1100)
verified 9404 bytes in 0.655824s (14.003 KiB/s)

Doing reset
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
````

However, if I add one more "nop" to the code, it will "crash" the core. So no more printing out. The first message after loading is this:
````
Open On-Chip Debugger 0.12.0-rc1
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
force hard breakpoints
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
9408 bytes written at address 0x00100000
downloaded 9408 bytes in 0.226568s (40.551 KiB/s)

Error: No working memory available. Specify -work-area-phys to target.
Warn : not enough working area available(requested 1100)
verified 9408 bytes in 0.607981s (15.111 KiB/s)

Doing reset
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
````
It seems fine in the output but actually it does not output anything. And after the second loading, openocd is unable to halt the core, seems that ibex does not take any "wfi". 
````
Open On-Chip Debugger 0.12.0-rc1
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
force hard breakpoints
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
Info : datacount=2 progbufsize=8
Error: unable to halt hart 0
Error:   dmcontrol=0x80000001
Error:   dmstatus =0x00000c82
Error: Fatal: Hart 0 failed to halt during examine()
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
````

I tried to increase the stack size (bigger than the whole program) in the linker file but no luck. 

Also, even if program for file size smaller than 9404B, it gives the error message of "No working memory available. Specify -work-area-phys to target." So how could I solve this?

medexs, 2024-10-05:
Hi, have you solved your issue yet?

marnovandermaas, 2025-04-11:
I'm not exactly sure what causes this. Would you mind reproducing this on the latest main branch?