lowRISC/ibex-demo-system #123: certain MEM_SIZE values cause failures

medexs, 2024-10-13:
I have successfully ported the Demo System to the Basys3 board (the branch in my forked repo is [here](https://github.com/lowRISC/ibex-demo-system/compare/main...medexs:ibex-demo-system:add-basys3-support)) and have found a bug when changing the value of the [MEM_SIZE](https://github.com/lowRISC/ibex-demo-system/blob/a1201cf2b99f8f4149c0971c04c655adbf1753c4/rtl/system/ibex_demo_system.sv#L41) parameter. If the value of MEM_SIZE is not a power of two, the core crashes due to some failure. It always happens when one of the jump instructions is executed. For every non power of two value of MEM_SIZE, there is one specific jump instruction that causes the failure after executing it. After executing the specific jump instruction, halting of the hart is not successfull and any communication with the board via openocd is not possible, so the bitstream has to be loaded again (this behaviour is similar to the one mentioned in issue #122 , but it was invoked differently).

Although I have been only testing it on the Basys3 board, I think that this issue will also be present on the Arty-A7 board.

nbdd0121, 2024-10-13:
The code was written assuming MEM_SIZE is power of two, e.g. the mask calculation relies on MEM_SIZE being power of two to work. Why do you need to be not a power of two?

medexs, 2024-10-13:
@nbdd0121 Thank you for a quick reply.

I am porting [NuttX](https://github.com/apache/nuttx) to Ibex and needed tu bump up the memory size from the original 64KiB and didn't set it to a power of two value because I didn't know that the implementation doesn't allow it. I spent a lot of time finding where the problem is as the NuttX port wasn't in a stable state yet so I didn't know if it's a SW or HW bug, because there are also methodology warnings in Vivado. Then I discovered that certain values of MEM_SIZE work as mentioned above. So I thought it's a bug.

Maybe there should be a comment or a note somewhere about this? So people don't have to poke into the implementation of the Xilinx primitives? Or is it possible to allow the size to be a non power of 2 value?

marnovandermaas, 2025-04-11:
@medexs feel free to open a PR with the comment that would have been useful to you.