lowRISC/ibex-demo-system #128: New IP integration (AES)

Inky-13, 2024-11-08:
I am currently trying to add an AES IP to this SoC. Is there any documentation on how to do this? Else, could I get some guidance as to where to start?

I am trying to use the IP from the [OpenTitan's project](https://github.com/lowRISC/opentitan/tree/master/hw/ip/aes) would this be the easiest way to do this. 

Any help would be appreciated.

marnovandermaas, 2024-11-11:
You can add a device to this enum: https://github.com/lowRISC/ibex-demo-system/blob/a1201cf2b99f8f4149c0971c04c655adbf1753c4/rtl/system/ibex_demo_system.sv#L93
You should also add it to the memory address space like the UART: https://github.com/lowRISC/ibex-demo-system/blob/a1201cf2b99f8f4149c0971c04c655adbf1753c4/rtl/system/ibex_demo_system.sv#L53-L55
You can use vendoring to vendor in the AES from OpenTitan, we already import some of them in this repo: https://github.com/lowRISC/ibex-demo-system/blob/main/vendor/lowrisc_ip.vendor.hjson
All of this will be a bit fiddly but you should be able to figure it out if you copy what you see in the rest of the repo. Once you've vendored things in you should be able to do everything else in `ibex_demo_system.sv`.

marnovandermaas, 2025-04-11:
Closing this due to inactivity, feel free to re-open if you have further enquiries.

closed 2025-04-11 by marnovandermaas