lowRISC/ibex-demo-system #130: Adding new Xilinx board for the demo

TarikHamedovic, 2024-12-12:
Hello 

I have been trying to add a Alinx AX7201 board to the demo and try out the Ibex processor but I have been having some trouble.
I have added my own `pins_ax7201.xdc` file in the `data` directory, aswell as a bare bones top module in `rtl/fpga` called `top_ax7201.sv`. Then I modified the `ibex_demo_system.core`  to include the aforementioned `ax720`1 files like this:
 ```
CAPI=2:
# Copyright lowRISC contributors.
# Licensed under the Apache License, Version 2.0, see LICENSE for details.
# SPDX-License-Identifier: Apache-2.0
name: "lowrisc:ibex:demo_system"
description: "Ibex Demo System for Arty A7 boards (both, -35 and -100)"
filesets:
  files_rtl:
    depend:
      - lowrisc:ibex:demo_system_core

  files_xilinx:
    depend:
      - lowrisc:ibex:rv_timer
      - lowrisc:ibex:fpga_xilinx_shared
    files:
      - rtl/fpga/top_artya7.sv
    file_type: systemVerilogSource
  
  files_ax7201:
    depend:
      - lowrisc:ibex:rv_timer
      - lowrisc:ibex:fpga_xilinx_shared
    files:
      - rtl/fpga/top_ax7201.sv
    file_type: systemVerilogSource
      
  files_xilinx_cw305:
    depend:
      - lowrisc:ibex:rv_timer
      - lowrisc:ibex:fpga_xilinx_shared
    files:
      - rtl/fpga/top_cw305.sv
    file_type: systemVerilogSource

  files_xilinx_cw312a35:
    depend:
      - lowrisc:ibex:rv_timer
      - lowrisc:ibex:fpga_xilinx_shared
    files:
      - rtl/fpga/top_cw312a35.sv
    file_type: systemVerilogSource

  files_sonata:
    depend:
      - lowrisc:ibex:rv_timer
      - lowrisc:ibex:fpga_xilinx_shared
    files:
      - rtl/fpga/top_sonata.sv
      - rtl/fpga/clkgen_sonata.sv
      - rtl/fpga/rst_ctrl.sv
    file_type: systemVerilogSource

  files_verilator:
    depend:
      - lowrisc:ibex:sim_shared
      - lowrisc:dv_verilator:memutil_verilator
      - lowrisc:dv_verilator:simutil_verilator
      - lowrisc:dv_verilator:ibex_pcounts
      - lowrisc:dv_dpi_c:uartdpi:0.1
      - lowrisc:dv_dpi_sv:uartdpi:0.1
    files:
      - dv/verilator/top_verilator.sv: { file_type: systemVerilogSource }
      - dv/verilator/ibex_demo_system.cc: { file_type: cppSource }
      - dv/verilator/ibex_demo_system.h:  { file_type: cppSource, is_include_file: true}
      - dv/verilator/ibex_demo_system_main.cc: { file_type: cppSource }
      - dv/verilator/demo_system_verilator_lint.vlt:  { file_type: vlt }

  files_constraints:
    files:
      - data/pins_artya7.xdc
    file_type: xdc
    
  files_constraints_ax7201:
    files:
      - data/pins_ax7201.xdc
    file_type: xdc

  files_constraints_cw305:
    files:
      - data/pins_cw305.xdc
    file_type: xdc

  files_constraints_cw312a35:
    files:
      - data/pins_cw312a35.xdc
    file_type: xdc

  files_constraints_sonata:
    files:
      - data/pins_sonata.xdc
    file_type: xdc

parameters:
  # XXX: This parameter needs to be absolute, or relative to the *.runs/synth_1
  # directory. It's best to pass it as absolute path when invoking fusesoc, e.g.
  # --SRAMInitFile=$PWD/sw/led/led.vmem
  # XXX: The VMEM file should be added to the sources of the Vivado project to
  # make the Vivado dependency tracking work. However this requires changes to
  # fusesoc first.
  SRAMInitFile:
    datatype: str
    description: SRAM initialization file in vmem hex format
    default: "../../../../../sw/c/build/blank/blank.vmem"
    paramtype: vlogparam

  # For value definition, please see ip/prim/rtl/prim_pkg.sv
  PRIM_DEFAULT_IMPL:
    datatype: str
    paramtype: vlogdefine
    description: Primitives implementation to use, e.g. "prim_pkg::ImplGeneric".

targets:
  default: &default_target
    filesets:
      - files_rtl
  synth:
    <<: *default_target
    default_tool: vivado
    filesets_append:
      - files_xilinx
      - files_constraints
    toplevel: top_artya7
    tools:
      vivado:
        part: "xc7a35tcsg324-1"  # Default to Arty A7-35
    parameters:
      - SRAMInitFile
      - PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx
    flags:
      use_bscane_tap: true
  synth_ax7201:
    <<: *default_target
    default_tool: vivado
    filesets_append:
      - files_ax7201
      - files_constraints_ax7201
    toplevel: top_ax7201
    tools:
      vivado:
        part: "xc7a200tfbg484-2"
    parameters:
      - SRAMInitFile
      - PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx
    flags:
      use_bscane_tap: true
  synth_cw305:
    <<: *default_target
    default_tool: vivado
    filesets_append:
      - files_xilinx_cw305
      - files_constraints_cw305
    toplevel: top_cw305
    tools:
      vivado:
        part: "xc7a100tftg256-2" # default to a100 part
        #part: "xc7a35tftg256-2" # a35 option
    parameters:
      - SRAMInitFile
      - PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx
    flags:
      use_bscane_tap: true
  synth_cw312a35:
    <<: *default_target
    default_tool: vivado
    filesets_append:
      - files_xilinx_cw312a35
      - files_constraints_cw312a35
    toplevel: top_cw312a35
    tools:
      vivado:
        part: "xc7a35tcsg324-1"
    parameters:
      - SRAMInitFile
      - PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx
    flags:
      use_bscane_tap: true
  synth_sonata:
    <<: *default_target
    default_tool: vivado
    filesets_append:
      - files_sonata
      - files_constraints_sonata
    toplevel: top_sonata
    tools:
      vivado:
        part: "xc7a50tcsg324-1"
    parameters:
      - SRAMInitFile
      - PRIM_DEFAULT_IMPL=prim_pkg::ImplXilinx

  sim:
    <<: *default_target
    default_tool: verilator
    filesets_append:
      - files_verilator
    toplevel: top_verilator
    tools:
      verilator:
        mode: cc
        verilator_options:
          # Disabling tracing reduces compile times but doesn't have a
          # huge influence on runtime performance.
          - '--trace'
          - '--trace-fst' # this requires -DVM_TRACE_FMT_FST in CFLAGS below!
          - '--trace-structs'
          - '--trace-params'
          - '--trace-max-array 1024'
          - '-CFLAGS "-std=c++14 -Wall -DVM_TRACE_FMT_FST -DTOPLEVEL_NAME=top_verilator"'
          - '-LDFLAGS "-pthread -lutil -lelf"'
          - "-Wall"
          - "-Wwarn-IMPERFECTSCH"
          # RAM primitives wider than 64bit (required for ECC) fail to build in
          # Verilator without increasing the unroll count (see Verilator#1266)
          - "--unroll-count 72"
    parameters:
      - PRIM_DEFAULT_IMPL=prim_pkg::ImplGeneric
```

When trying to build the bitstream using `fusesoc` with this command: 
```
fusesoc --cores-root=. run --target=synth_ax7201 --tool=vivado --setup --build lowrisc:ibex:demo_system --no-cache --verbose
```
I get the following error:
```
ERROR: Failed to determine work root. Could not resolve target
```
Or sometimes aswell: 
```
ERROR: No tool was supplied on command line or found in 'lowrisc:ibex:demo_system' core description

```

Am I missing something?

Thank you in advance.

marnovandermaas, 2024-12-13:
I've never used the `tool` option in the demo system, this is the command I usually use:
`fusesoc --cores-root=. run --target=synth --setup --build lowrisc:ibex:demo_system`

The default tool for synth is Vivado, which is the same for your ax7201. Did you have any trouble building the original bitstream for the arty a7? Then we at least know that the problem is with your added changes as opposed to your dev environment.

TarikHamedovic, 2024-12-13:
The original bitstream for the ArtyA7 built without a problem, I aswell tried the other synth options. I used the python venv. I originally tried without the tool option aswell and got the same error.

marnovandermaas, 2024-12-13:
I found this old issue: https://github.com/olofk/fusesoc/issues/586
Any further logging from fusesoc?

TarikHamedovic, 2024-12-14:
Here is a [gist ](https://gist.github.com/TarikHamedovic/8eaccdf5fab059c0c0669f66528eef2b)with all of the files I have added.

Using `--verbose' gives me back the same ERROR messages. When I don't use `--tool=vivado' I get:
```
ERROR: No tool was supplied on command line or found in 'lowrisc:ibex:demo_system' core description
```

When I use `--tool=vivado`  I get:
```
ERROR: Failed to determine work root. Could not resolve target
```

marnovandermaas, 2025-04-11:
Since this issue has been opened there have been a few more boards added to the demo system. Would you mind having a look at those? I cannot open that gist.

This is the command I would try:
```sh
fusesoc --cores-root=. run --target=synth_ax7201 --setup --build lowrisc:ibex:demo_system
```