lowRISC/ibex-demo-system #138: Unable to load "Hello World " demo application to the programmed FPGA (Boolean Board )

Prathmesh-shalu141, 2025-03-29:
I am using a Boolean board for programming and loading the "Hello World " demo application. I have successfully programmed the FPGA, but when I am loading the demo application, it is throwing an error as follows:
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
**cselab@cselab-Precision-3660:~/IBEX/ibex-demo-system$**  ./util/load_demo_system.sh run ./sw/c/build/demo/hello_world/demo

xPack Open On-Chip Debugger 0.12.0+dev-01312-g18281b0c4-dirty (2023-09-04-22:31)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
force hard breakpoints
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x13631093 (mfg: 0x049 (Xilinx), part: 0x3631, ver: 0x1)
**Error: dtmcontrol is 0. Check JTAG connectivity/board power.**
Warn : target riscv.cpu examination failed
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
**Error: Target not examined yet**

marnovandermaas, 2025-04-07:
I think this issue will be solved once we close: https://github.com/lowRISC/ibex-demo-system/issues/116

marnovandermaas, 2025-04-08:
@Prathmesh-shalu141 I've now merged a PR that should hopefully fix this. Can you double check?

Prathmesh-shalu141, 2025-04-11:
Yes,, it is functioning. However, when I run the simulation now, it fails with the error that is attached below.

[image]

marnovandermaas, 2025-04-11:
Are you building the simulator with the following command?
```sh
fusesoc --cores-root=. run --target=sim --tool=verilator --setup --build lowrisc:ibex:demo_system
```

Prathmesh-shalu141, 2025-04-11:
Yes

marnovandermaas, 2025-04-11:
Have you tried removing the build directory and rebuilding? Also can you tell me the git hash that you're running on top of?

Prathmesh-shalu141, 2025-04-13:
Yes,, I did that, but the error is still occurring. 

The git hash that i am using is : ed9f04072c02a48e9470c4219607a865373f5042

Prathmesh-shalu141, 2025-04-14:
cselab@cselab-Precision-3660:~/IBEX/ibex-demo-system$ verilator --version
Verilator 5.035 devel rev v5.034-93-ged1e3ec4a

marnovandermaas, 2025-04-14:
Would you mind trying with v5.024? If that solves your issue then I can see whether we can bump Verilator versions for the demo system.

Prathmesh-shalu141, 2025-04-15:
Okay, I will check it. For now you can close this issue.

closed 2025-04-22 by marnovandermaas