lowRISC/ibex-demo-system #140: Unable to load a application if the size is bigger than a certain value (Boolean Board)

Prathmesh-shalu141, 2025-04-22:
If my program size is greater than 9408 bytes, it's not working. Please check the screenshot shot attached below.

[image]

marnovandermaas, 2025-04-22:
I've seen the "No working memory available" error before but it looks like the program is verified after being programmed, so I think this might have succeeded. Can you actually connect GDB and see what it's doing?

Prathmesh-shalu141, 2025-04-24:
can you please explain it in details so that i will proceed further.

marnovandermaas, 2025-04-24:
Here's how you can connect GDB: https://github.com/lowRISC/ibex-demo-system?tab=readme-ov-file#debugging-an-application

What you need to do to debug depends on what your program does. I would make your program blink an LED to check whether the programming was successful or not.