lowRISC/ibex-demo-system #146: Use of variables before declaration in spi_host.sv commit 1e5da83

Soeren-Petersen, 2025-08-28:
In ibex-demo-system/rtl/system/spi_host.sv I have a problem compiling with Cadence Xcelium 24.09-s008.
The  
65: typedef enum logic[1:0] {
66:     IDLE,
67:     START,
68:     SEND,
69:     STOP
70:  } spi_state_t;
and
75:  spi_state_t state_q, state_d;

comes after (line 65 to 75) the first use of them (line 35)

34:  logic sck_en;
35: assign sck_en = (state_q == SEND);

Is that allowed in other tools?

Soeren-Petersen, 2025-09-03:
> EXTERNAL EMAIL - This email was sent by a person from outside your organization. Exercise caution when clicking links, opening attachments or taking further action, before validating its authenticity.
> Secured by Check Point <https://www.checkpoint.com/harmony/email-security/email-office?friew>
>
>
> Yes, that is allowed in other tools, but we should fix this. @Soeren-Petersen<https://github.com/Soeren-Petersen> would you mind opening a PR to put the definition of spi_state_t above its first use?

Hello Marno,

I have tried to make the PR. It is my very first time using Github, so I hope it is done correctly.
Best regards
Søren

marnovandermaas, 2025-11-04:
Fixed in: https://github.com/lowRISC/ibex-demo-system/pull/148

closed 2025-11-04 by marnovandermaas